PX4/PX4-containers #2: Imrove ROS Docker container, various things

AndreasAntener, 2015-02-24:
**TODO:**
  * use https://docs.docker.com/articles/using_supervisord/
  * configure ssh to work with Vagrant out of the box
  * check if unprivileged execution is possible and still easy to use

dagar, 2017-01-07:
You mainly want ssh?

AndreasAntener, 2017-01-07:
Ugh, this issue is a bit stale I guess. What I was trying to do in the beginning is have Vagrant as user interface and use Docker underneath that. At that time it turned out to be even more complicated and lead to even more problems. This would still make sense for companies as it gives them a complete environment starting at the OS level. Lets close this issue. A fresh look is needed anyway if someone wants to have another go at Vagrant (and the likes).

closed 2017-01-07 by AndreasAntener