PX4/PX4-containers #23: Structuring Docker containers

AndreasAntener, 2016-06-06:
@dagar @julianoes @LorenzMeier  the container hierarchy topic. My suggestion:

- base (determines latest supported OS plus base apps if any, main thing is we only set the OS once)
  - dev (px4 toolchain)
    - sitl
    - sitl-ros
  - snappy
  - raspberry
  - more?

Furthermore I'd suggest once we have it setup we do a one-off build with a 14.04 base (tag the images) and then switch to 16.04. The same with ROS Jade and Gazebo 6, then switch to Kinetic and Gazebo 7. If we're really crazy later we can start generating the Docker files in a matrix to support various envs but I don't think it's worth it anytime soon.

Questions
- do we need to separate Snappy/RPI toolchains from Dev?
- do we need SITL/ROS on top of the other toolchain containers or could we have them just inherit from base?

julianoes, 2016-06-06:
@AndreasAntener: The RPi2 should be fairly small and simple but the Snappy one needs SDKs pulled in that are not public, see https://github.com/PX4/containers/pull/22/files.

I would like the base to be more minimal and not to include anything NuttX / arm toolchain related. Basically, I would make it to just what is needed to do `make posix`.

AndreasAntener, 2016-06-06:
'make posix', would that include jMAVSim?

AndreasAntener, 2016-06-06:
Updated structure suggestion in that case

julianoes, 2016-06-06:
> 'make posix', would that include jMAVSim?

No, I'd say minimal, without gazebo/jMAVSim, etc.

AndreasAntener, 2016-06-06:
What's the purpose of that container then?

julianoes, 2016-06-06:
Build POSIX, but yer you're right, might as well put more in there.

AndreasAntener, 2016-06-06:
Trying to figure out what we should actually split out, too much doesn't make sense if we have 1 test command that builds/runs every platform.

Also, if we want something very specific for only running unit-tests we should look into minimal base images (like Alpine)

dagar, 2016-06-06:
I think this makes sense, but we'll have to see if maintaining all these different environments is worth the effort. It's nice to have small images for a build system that has to fetch the full thing every single time. For my own workstation I don't care if my container has nuttx, raspi, gazebo, ros, etc altogether as long as there aren't conflicts. 

Initially I wanted one command `make check` that builds everything we care about and runs some testing, but there's just too much. It's looking like things will end up split out over travis-ci, semaphore-ci, possibly circle-ci and jenkins.

Travis-ci could just compile + upload the nuttx firmware binaries.
Semaphore-ci could build SITL, run unit tests, and do some mission testing.
We'll have to decide where snappy fits.

mcharleb, 2016-06-06:
Would you be open to passing a EULA acceptance flag for the Hexagon SDK when running the docker image? Minecraft images do this:

https://www.ctl.io/developers/blog/post/deploy-a-minecraft-server-using-docker-and-blueprints

docker run -i -t -p=25565:25565 -e EULA=true kitematic/minecraft

dagar, 2016-06-06:
@mcharleb if a user of the docker image is required to agree to the EULA could it be made more widely available?

AndreasAntener, 2016-06-08:
@dagar ok. One large group of users probably needs the base plus Nuttx, the other group Snappy. That split makes a lot of sense at the moment. Linux support (RPi) could possibly also go into the Nuttx container, maybe give it a different name then (I'm inclined to say "default" but that's kind of discriminating the others ;))

dagar, 2016-06-08:
We're close to having nuttx, sitl, snappy building out of docker containers on travis-ci and semaphoreci. I'll revisit the hierarchy and organization once things settle.

mcharleb, 2016-06-09:
@dagar That would be the intention. It would be fully redistributable and ready to run with the SDK preinstalled if the EULA was accepted. I'm socializing the idea with the Hexagon SDK team.

dagar, 2016-06-22:
I'll start doing some of this now.

- base (determines latest supported OS plus base apps, 'make posix' and jMAVSim)
  - nuttx
  - snappy
  - raspberry
  - gazebo
  - gazebo-ros

dagar, 2016-12-04:
https://github.com/PX4/containers/pull/39

dagar, 2016-12-28:
Done!

closed 2016-12-28 by dagar