PX4/PX4-containers #30: Snapdragon add ninja-build

dagar, 2016-06-22:
@julianoes could you add ninja-build to the snapdragon image? That should help with build times.
https://github.com/PX4/containers/blob/master/docker/px4-dev/DockerfileBase#L25

https://github.com/PX4/Firmware/issues/4871

dagar, 2016-06-22:
On second thought let's use this as an excuse to structure the containers. I'll strip nuttx out of px4-dev-base and create px4-dev-nuttx. Then you can base snappy off of px4-dev-base and they'll have most of the same dependencies.
Standby.

julianoes, 2016-06-23:
Up to you, ok.

dagar, 2016-06-25:
@julianoes the next time you build px4-snappy you should consider squashing before pushing to docker hub. Removing the extra layers (from each ADD, RUN, etc) might save some space.
https://github.com/jwilder/docker-squash

julianoes, 2016-07-07:
@dagar: What's the state here, should I still add ninja-build and do a squashed build?

dagar, 2016-07-07:
If it's trivial for you to add a couple packages, rebuild, and push an updated px4-snappy then you should go ahead. Add ninja-build and ccache and your snappy build will get a bit faster.

I have a partially complete branch where I rearranged all the Dockerfiles into a hierarchy all built off of a single tiny base image that can build SITL. The problem is the current images are now intertwined with several different build systems and nothing is tagged. So I'm contemplating either going through and tagging all the images, updating the build systems, etc OR just changing everything in one swoop.

julianoes, 2016-07-13:
@dagar: I've added ninja-build and ccache and verified that it still builds.
I'm pushing it as julianoes/px4-snappy right now.

Unfortunately, the docker-squash did not work. I followed the [usage instructions](https://github.com/jwilder/docker-squash#usage) but it did not create a new docker image. When I do `docker images` I couldn't see the loaded docker image, even when I gave it a different tag.

dagar, 2016-07-13:
Thanks. Too bad the squash didn't work.

Can you tag your image? YYYY-MM-DD is probably sufficient. I'm going to go through and tag the rest, update the build systems, then open a PR my Dockerfile hierarchy change.

julianoes, 2016-07-14:
@dagar retagged and pushed as `docker.io/julianoes/px4-snappy-2016-07-13`.

dagar, 2016-07-15:
Thanks @julianoes 
It looks like the only way to tag automated docker builds is to branch the associated repo.

dagar, 2016-07-15:
@julianoes did you create an entire new docker repo? You should be able to tag the existing one in it's current state, and then access it as `julianoes/px4-snappy:2016-07-13`.

julianoes, 2016-07-15:
I tagged it and pushed it to my docker. Did I do something wrong?

dagar, 2016-07-15:
Unless I misunderstand something about docker here (quite possible) didn't you create a different image named `px4-snappy-2016-07-13` rather than creating a tag within px4-snappy?

julianoes, 2016-07-16:
I used -t to tag it, not sure.

dagar, 2016-07-16:
I guess both are fine.

closed 2016-07-16 by dagar

dagar, 2016-07-16:
I've switched semaphore-ci to use px4-snappy-2016-07-13.