PX4/PX4-containers #36: Gcc 4.9.3 needed for nuttx_v3_xxxxxx builds on CI - keep losing it

davids5, 2016-10-20:
@dagar 

I have strange results 
https://semaphoreci.com/px4/firmware/branches/nuttx_v3_sans_mavstation/builds/1

I build and pass nuttx_v3 on all CI, I delete mavsataion on a branch of the nuttx_v3 and semaphoreci  fails fmuv2 code space overflow. 

Where is the container chosen and where is the compiler set for semaphoreci?

dagar, 2016-10-20:
Right, so for nuttx_v3 you've changed travis-ci to GCC 4.9, but because the semaphoreci config isn't stored in tree you're stuck with the main configuration.

That certainly sucks. In this particular situation we could just switch semaphoreci to GCC 4.9. Semaphoreci builds all the targets travis-ci does plus some lower priority extras (discovery).

Does that sound acceptable?

LorenzMeier, 2016-10-20:
@dagar Let's do that.

davids5, 2016-10-20:
@dagar 

Daniel to the rescue!

I think it does and does not - but you tell me.  I am thinking of the others.....
semaphoreci builds check - with 4.9.3 things builds will fit that will not fit on 4.8.4 - So does that mean that we  get an early false positive from one CI and then later a fail? 

Can we just install both compilers and have a way in the makefile to tell the it which compiler?

LorenzMeier, 2016-10-20:
@davids5 We are actually using 5.x on Ubuntu / Arch and 4.9 on Mac OS: https://github.com/PX4/homebrew-px4/blob/master/gcc-arm-none-eabi.rb

So really just Travis is stuck with 4.8. Let's move everything to 4.9.

dagar, 2016-10-20:
Yes, but only the ones built on both.
px4fmu-v1, v2, v4, mindpx, tap

In my experience trying to handle multiple compiler versions within the Makefile gets awkward.

dagar, 2016-10-20:
Just saw @LorenzMeier's comment. Switch to 4.9.3 everywhere?

Semaphoreci has been updated.

LorenzMeier, 2016-10-20:
Everywhere please.

dagar, 2016-10-20:
Maybe you missed it, I edited the comment. Isn't GDB in 4.9 a problem?

dagar, 2016-10-20:
https://github.com/PX4/containers/pull/37

dagar, 2016-10-21:
Main builds (uploaded for QGC) updated to GCC 4.9.3, and circleci building with GCC 5.4 (make quick_check).

closed 2016-10-21 by dagar