PX4/PX4-containers #38: DockerfileSnapdragon too big for semaphoreci

dagar, 2016-11-25:
The snappy build on semaphoreci was failing due to the size of the docker image. Unfortunately semaphoreci doesn't keep build history for deleted branches.

It's been deleted from the PX4 semaphore configuration for now.

    docker-cache restore
    docker pull julianoes/px4-snappy-2016-10-16
    docker run --rm -v `pwd`:`pwd`:rw -v $SEMAPHORE_CACHE_DIR/ccache:$HOME/.ccache:rw -e CCACHE_DIR=$HOME/.ccache -e CCACHE_SLOPPINESS=time_macros -w=`pwd` --user=root -it julianoes/px4-snappy-2016-10-16 /bin/bash -c "make eagle_default"

FYI @julianoes @mcharleb

julianoes, 2016-11-25:
I was trying to create a smaller docker image but I'm currently blocked by https://github.com/ATLFlight/cross_toolchain/pull/10.

mcharleb, 2016-11-28:
@julianoes Merged, sorry didn't see the PR.

mcharleb, 2016-11-28:
@julianoes @dagar Let me know what I can do to make this work. @julianoes do you need a script to reduce the hexagon SDK to a bare minimum?

dagar, 2016-11-28:
Ping me when it's time to re-enable semaphore.

julianoes, 2016-11-29:
@mcharleb the problem is that the cmake gcc check suddenly uses target/hexagon/lib/v60 instead of v50, so the `-trim` option for `installv3.sh` can't be applied anymore, thus leading to a bigger image.

mcharleb, 2016-11-29:
@julianoes I can fix the cmake gcc check.

mcharleb, 2016-11-29:
@julianoes I updated installv3.sh and removed more files with the -trim option. It seems to work for me. Let me know if you have any issues

mcharleb, 2016-11-30:
@julianoes You will also need an updated cmake_hexagon: https://github.com/ATLFlight/cmake_hexagon/commit/a55fb8e306b06a4d3ab7f46536904b045923f46a

dagar, 2016-12-28:
docker pull lorenzmeier/px4-dev-snapdragon
    docker run --rm -v `pwd`:`pwd`:rw -e CI=true -w=`pwd` -e LOCAL_USER_ID=$UID -it lorenzmeier/px4-dev-snapdragon /bin/bash -c "make eagle_default"

mcharleb, 2016-12-28:
@dagar @julianoes @LorenzMeier I created docker images for Snapdragon Flight and buid tests: https://github.com/ATLFlight/containers

You still need access the the Hexagon SDK and qrlSDK to create them.

dagar, 2017-01-02:
Solved with minimalSDK.tgz

closed 2017-01-02 by dagar