PX4/PX4-containers #40: PX4Flow build is broken in latest px4io/px4-dev-ros image

Alexey-Kamenev, 2016-12-07:
[PX4Flow](https://github.com/PX4/Flow) build is currently broken in ```px4io/px4-dev-ros``` image in case you do ```docker pull px4io/px4-dev-ros```. I tried building the image myself using the latest sources but ```docker build``` failed with the same error as [here](https://hub.docker.com/r/px4io/px4-dev-ros/builds/bsyqmnk5pztuzksupo95dxy/).
I ended up taking [this](https://hub.docker.com/r/px4io/px4-dev-ros/builds/bwjzbdnwgbnydnpjbp9vc3g/) image and was able to build PX4Flow just fine.

dagar, 2017-01-03:
@Alexey-Kamenev could you try again with px4io/px4-dev-ros? We've pushed an updated px4-dev-ros now with Ubuntu 16.04, Gazebo 7, and ROS Kinetic.

dagar, 2017-01-27:
@Alexey-Kamenev any update?

closed 2017-02-22 by Alexey-Kamenev