PX4/PX4-containers #51: Default ccache directory is owned by root

AndreasAntener, 2017-01-07:
Prevents compilation as normal user.
Having to specify ` -e CCACHE_DIR=${HOME}/.ccache \ ` doesn't make sense to me. @dagar is it just owned by root because everything gets installed by root during the container build?

dagar, 2017-01-07:
This is so docker stores the cache outside of the container and seemed like as good a place as any to keep it.

The root ownership happens if the shared volume (docker -v) doesn't exist when you run. I didn't hit it because all of my machines already had a .ccache.

Can you try the 2017-01-06 tag? I believe I fixed the ccache root ownership issue.

AndreasAntener, 2017-01-07:
Ok I see now what you tried to do. I was talking about the ccache directory inside the container at /tmp/.ccache which is owend by root and used by default if you enter the container.

Mapping the cache outside the container defies the purpose of having a container IMHO. As long as my container runs it has it's state. Every new container (run) should get a new state. Otherwise I'll restart an old one.

dagar, 2017-01-07:
Oh, it's the tmp directory causing problems? I added that without much thought. Should we drop it?

I agree in principle, but a lot of this evolved trying to squeeze more out of these free CI services and the use case is a bit different. On travis-ci each build starts on a fresh machine, clones the repo, pulls our docker image, builds, and then throws it all away. However you can bundle up the .ccache directory and restore it the next build.

The cached results in ccache depend on the particular compiler binary. There's really no harm in letting it mix in with the hosts cache.

dagar, 2017-01-07:
TMPDIR removed 8e72cea

@AndreasAntener could you now try tag 2017-01-07?

commit 8e72cea7 (2017-01-07): px4-dev-base remove ccache tmpdir

 - closes #51
closed 2017-01-07 by dagar