PX4/PX4-containers #58: Coverity image broken

LorenzMeier, 2017-01-14:
@dagar The image in use for Coverity in Travis is broken:

```
[  3%] Generating uORB topic headers
python import error:  No module named em
Required python packages not installed.
On a Debian/Ubuntu system please run:
  sudo apt-get install python-empy
  sudo pip install catkin_pkg
On MacOS please run:
  sudo pip install empy catkin_pkg
On Windows please run:
  easy_install empy catkin_pkg
make[3]: *** [src/modules/uORB/topics/actuator_armed.h] Error 1
make[2]: *** [CMakeFiles/msg_gen.dir/all] Error 2
make[2]: *** Waiting for unfinished jobs....
[  3%] Building CXX object 
```

LorenzMeier, 2017-01-14:
Its not really the image. I'm working on a workaround.

closed 2017-01-14 by LorenzMeier

dagar, 2017-01-14:
Yes, it turned out to be far easier to let the travis-ci coverity addon auto install and work as intended.