PX4/PX4-containers #67: Unmappable characters when building jMAVSim

martinlowinski, 2017-07-28:
Hi,

after [setting up](https://dev.px4.io/en/test_and_ci/docker.html) the `px4io/px4-dev-ros` docker image on a clean Ubuntu 17.04 running in a VM, I get errors that there are unmappable characters for the given encoding from the java compiler.

The container is created like this:

```
xhost +
docker run -it -d --privileged \
    -v /home/$USER/Firmware:/job:rw \
    -v /tmp/.X11-unix:/tmp/.X11-unix:ro \
    -e DISPLAY=:0 \
    -p 14556:14556/udp \
    --name=px4-dev-ros px4io/px4-dev-ros /bin/bash
```

jMAVSim was build using `make posix_sitl_default jmavsim` which produces said unmappable character errors in the jMAVSim source code.

According to [stackoverflow](https://stackoverflow.com/questions/26067350/unmappable-character-for-encoding-ascii-but-my-files-are-in-utf-8), this is an encoding issue that can be fixed by running `export JAVA_TOOL_OPTIONS=-Dfile.encoding=UTF8` before make. Does it make sense to configure the source code encoding in the dockerfile, e.g. via `ENV`?

dagar, 2017-08-01:
Can you try it within the container to verify it solves the problem?

martinlowinski, 2017-08-01:
Yes. I added `ENV JAVA_TOOL_OPTIONS=-Dfile.encoding=UTF8` at the end of `containers/docker/px4-dev/Dockerfile_simulation`, built the image, started the container and executed `make posix_sitl_default jmavsim`:

```
Picked up JAVA_TOOL_OPTIONS: -Dfile.encoding=UTF8
Buildfile: /job/Tools/jMAVSim/build.xml

make_dirs:

compile:
    [javac] Compiling 73 source files to /job/Tools/jMAVSim/out/production/jMAVSim
    [javac] Note: /job/Tools/jMAVSim/src/me/drton/jmavsim/PeripherialBuzzer.java uses unchecked or unsafe operations.
    [javac] Note: Recompile with -Xlint:unchecked for details.

create_run_jar:
      [jar] Building jar: /job/Tools/jMAVSim/out/production/jmavsim_run.jar

copy_res:
     [copy] Copying 14 files to /job/Tools/jMAVSim/out/production

BUILD SUCCESSFUL
```

It picks up the encoding options and compiles jMAVSim successfully.

Otherwise, this is the error without setting the encoding:

```
Buildfile: /job/Tools/jMAVSim/build.xml

make_dirs:
    [mkdir] Created dir: /job/Tools/jMAVSim/out/production/jMAVSim

compile:
    [javac] Compiling 73 source files to /job/Tools/jMAVSim/out/production/jMAVSim
    [javac] /job/Tools/jMAVSim/src/me/drton/jmavsim/SimpleEnvironment.java:14: error: unmappable character for encoding ASCII
    [javac]     public static final double M = 0.0289644;  // molar mass of Earth???s air [kg/mol]
    [javac]                                                                      ^
    [javac] /job/Tools/jMAVSim/src/me/drton/jmavsim/SimpleEnvironment.java:14: error: unmappable character for encoding ASCII
    [javac]     public static final double M = 0.0289644;  // molar mass of Earth???s air [kg/mol]
    [javac]                                                                       ^
    [javac] /job/Tools/jMAVSim/src/me/drton/jmavsim/SimpleEnvironment.java:14: error: unmappable character for encoding ASCII
    [javac]     public static final double M = 0.0289644;  // molar mass of Earth???s air [kg/mol]
    [javac]                                                                        ^
    [javac] 3 errors

BUILD FAILED
```

dagar, 2017-08-01:
Great, if you open a PR I'll get it merged asap.

PR #68 (2017-08-02): Use UTF8 encoding in java tools
Fixes #67
closed 2017-08-02 by dagar