PX4/PX4-containers #81: Docker container - issue without permission errors

margaridaCF, 2017-11-08:
Hi!

   I saw that you have another issue with docker containers but as I do not get any permission errors I opened another issue. I actually get to the point where gazebo opens for a brief period and then shuts down. I cannot even find the error message...

   Here are my installation steps. I'm using Ununtu Xenial and followed docker's instructions for installation.
   1. create a folder inside home for the git clone of the Firmware (this might seem obvious to you but as a first time user of both docker and Px4, it was not at all obvious to me).
```
cd ~
mkdir docker_px4_gazebo
cd docker_px4_gazebo
git clone https://github.com/PX4/Firmware.git
```

2. Get the docker part. I get around the permission issue (I think) by exiting and then entering as root.
```
docker run -it --privileged -v `pwd`:`pwd`:rw -v /tmp/.X11-unix:/tmp/.X11-unix:ro -e DISPLAY=:0 -p 14556:14556/udp --name=px4_gazebo px4io/px4-dev-ros:2017-05-31 bash
exit
docker exec -u 0 -it px4_gazebo bash
make posix_sitl_default gazebo
```
I can see the Gazebo window flash by, and it's gone.
Here is the output [dockerGazebo_output.txt](https://github.com/PX4/containers/files/1454517/dockerGazebo_output.txt)
Thanks for any feedback.

margaridaCF, 2017-11-09:
Hi!

   Just figured it out. Need to do xhost + every time.

  Thanks anyway.

closed 2017-11-09 by margaridaCF

mymultiverse, 2018-03-20:
@margaridaCF can you help me out [here](https://github.com/PX4/containers/issues/108) I am getting issues with xhost +