PX4/PX4-containers #105: Running PX4 containers on raspberry pi 3 caused "exec format error"

jedichen121, 2018-01-17:
I tried to run the px4 autopilot in a container on raspberry pi 3. However, every time I tried to build it shows 
`standard_init_linux.go:195: exec user process caused "exec format error"
`
I installed docker on raspbian using the convenience script. I then followed steps described in this page. https://dev.px4.io/en/test_and_ci/docker.html

I've tried different make command, such as `make posix_rpi_native`, `make posix_sitl_default`, and `make posix_sitl_default gazebo`, but all gave the same results. 

I thought it might be caused by the OS I'm running. So I've used raspbian stretch lite download from raspberrypi.com, two versions of Emlid image (stretch and jessie) from emlid website. But none of these works. 

Below are some information that might help debug. 

The output of the command is:
```
guessing PX4_DOCKER_REPO based on input
PX4_DOCKER_REPO: px4io/px4-dev-raspi:2017-12-30
standard_init_linux.go:195: exec user process caused "exec format error"
```

Running `docker images` it shows the correct container has been downloaded:

```
REPOSITORY                TAG                 IMAGE ID            CREATED             SIZE
px4io/px4-dev-base        latest              0787831f3c53        12 days ago         695MB
px4io/px4-dev-raspi       2017-12-30          67f37542d7bf        2 weeks ago         969MB
python                    2.7-slim            957faace74fa        4 weeks ago         124MB
hello-world               latest              75280d40a50b        7 weeks ago         1.69kB
px4io/px4-dev-nuttx       2017-10-23          eb7bfffbcfb0        2 months ago        1.17GB
```

**Output of `docker version`:**

```
Client:
 Version:	18.01.0-ce
 API version:	1.35
 Go version:	go1.9.2
 Git commit:	03596f5
 Built:	Wed Jan 10 20:25:15 2018
 OS/Arch:	linux/arm
 Experimental:	false
 Orchestrator:	swarm

Server:
 Engine:
  Version:	18.01.0-ce
  API version:	1.35 (minimum version 1.12)
  Go version:	go1.9.2
  Git commit:	03596f5
  Built:	Wed Jan 10 20:21:22 2018
  OS/Arch:	linux/arm
  Experimental:	false
```

**Output of `docker info`:**

```
Containers: 6
 Running: 0
 Paused: 0
 Stopped: 6
Images: 11
Server Version: 18.01.0-ce
Storage Driver: overlay2
 Backing Filesystem: extfs
 Supports d_type: true
 Native Overlay Diff: true
Logging Driver: json-file
Cgroup Driver: cgroupfs
Plugins:
 Volume: local
 Network: bridge host macvlan null overlay
 Log: awslogs fluentd gcplogs gelf journald json-file logentries splunk syslog
Swarm: inactive
Runtimes: runc
Default Runtime: runc
Init Binary: docker-init
containerd version: 89623f28b87a6004d4b785663257362d1658a729
runc version: b2567b37d7b75eb4cf325b77297b140ea686ce8f
init version: 949e6fa
Security Options:
 seccomp
  Profile: default
Kernel Version: 4.9.65-rt57-v7+
Operating System: Raspbian GNU/Linux 9 (stretch)
OSType: linux
Architecture: armv7l
CPUs: 4
Total Memory: 927.3MiB
Name: raspberrypi
ID: L2SN:3DEE:FHCX:K3SA:DMHD:3KLR:Z4TE:UB4R:GMZM:VYTW:STDS:LOPG
Docker Root Dir: /var/lib/docker
Debug Mode (client): false
Debug Mode (server): false
Username: jedichen121
Registry: https://index.docker.io/v1/
Labels:
Experimental: false
Insecure Registries:
 127.0.0.0/8
Live Restore Enabled: false

WARNING: No memory limit support
WARNING: No swap limit support
WARNING: No kernel memory limit support
WARNING: No oom kill disable support
WARNING: No cpu cfs quota support
WARNING: No cpu cfs period support
```

dagar, 2018-08-04:
Sorry I missed this issue. Can you confirm if this is still a problem?

jedichen121, 2018-08-13:
Hi, I just tried it again with the latest px4 code. The issue is still there. Same old error. Maybe this some tools used does not support arm v7?

dagar, 2018-08-14:
Yes these containers are built with desktop use in mind. If you're interested we could try adding an arm version as well. I don't know how involved this will be.

I would start with px4-dev-base (https://github.com/PX4/containers/blob/master/docker/px4-dev/Dockerfile_base) and try to build it directly on your raspberrypi.

dagar, 2018-08-14:
Here's Ubuntu armhf - https://hub.docker.com/r/armhf/ubuntu/

closed 2018-11-25 by dagar