PX4/PX4-containers #106: PX4Flow build is broken in latest px4io/px4-dev-ros image

vijay609, 2018-01-18:
[PX4Flow](https://github.com/PX4/Flow) build is broken in the [latest](https://hub.docker.com/r/px4io/px4-dev-ros/builds/bsdgmqwpaos6mo3ldwuoerd/) `px4io/px4-dev-ros` image. Here's the output of make :

> user@82f375162c2c:/src/px4flow/Flow$ make px4flow-v2_default
Checked mavlink submodule, correct version found
%%%%
%%%% Building px4flow-v2_default on baremetal in /src/px4flow/Flow/Build/px4flow-v2_default.build/
%%%%
make[1]: Entering directory '/src/px4flow/Flow/Build/px4flow-v2_default.build'
%  PX4_BASE            = /src/px4flow/Flow/
%  GIT_DESC            = 1bb4e50d305ebd952851e8cbea4b3312e9744360
%  BUILD_URI           = "localhost"
%  CONFIG              = px4flow-v2_default
%  PX4_TARGET_OS       = baremetal
**make[1]: arm-none-eabi-gcc: Command not found**
**make[1]: arm-none-eabi-gcc: Command not found**
__/src/px4flow/Flow/makefiles/baremetal/toolchain_gnu-arm-eabi.mk:57: *** Unsupported version of arm-none-eabi-gcc, found:  instead of one in: 4.7.4 4.7.5 4.7.6 4.8.4 4.9.3 5.4.1.  Stop.__
make[1]: Leaving directory '/src/px4flow/Flow/Build/px4flow-v2_default.build'
/src/px4flow/Flow/makefiles/baremetal/firmware_baremetal.mk:30: recipe for target '/src/px4flow/Flow/Build/px4flow-v2_default.build/firmware.px4' failed
make: *** [/src/px4flow/Flow/Build/px4flow-v2_default.build/firmware.px4] Error 2

I tried to pull the docker image that @Alexey-Kamenev referenced in this issue #40 and was able to build the `px4flow` firmware successfully.
`docker pull px4io/px4-dev-ros@sha256:b9c4f3c4c1d021db42815a9eecd8e65fc72008033eec83cd7608e6c6c43601d7`

dagar, 2018-01-18:
I see at least 2 issues.

First of all the compilers are better split now (to keep image size down). You'll need px4io/px4-dev-nuttx for the arm-none-eabi-gcc compiler. The next issue is that the latest px4io/px4-dev-nuttx has been upgraded to GCC 7. This may work after modifying `Flow/makefiles/baremetal/toolchain_gnu-arm-eabi.mk` to tolerate that version, or you may need to make changes.

As a short term solution you could use any of the other tags to build. https://hub.docker.com/r/px4io/px4-dev-nuttx/tags/. I believe anything before Dec 30, 2017 should be GCC 5.4.

closed 2018-01-18 by dagar

vijay609, 2018-01-18:
@dagar,
First of all __Thank You__ for the quick reply.
I had tried the latest `px4io/px4-dev-nuttx` image prior to posting the issue but got the compiler version mismatch error like you pointed out.
I tried the older `px4io/px4-dev-nuttx` image (_tag 2017-12-08_) like you suggested and am able to build the firmware now.