PX4/PX4-containers #108: bash: xhost: command not found

mymultiverse, 2018-03-20:
I am trying to run xhost + inside the container but I am getting this error. 

```
~/src/Firmware$ sudo ./Tools/docker_run.sh 'bash'
root@9c1395be4cae:/home/yograj/src/Firmware# xhost + 
bash: xhost: command not found
```

lamping7, 2018-03-20:
What are trying to do here? You shouldn't be doing xhost inside the container if you are just trying to get gui apps to show. What `run` command did you use to create the container?

mymultiverse, 2018-03-21:
I am trying to run with 
```
sudo docker run -it --privileged \
--env=LOCAL_USER_ID="$(id -u)" \
-v ~/src/Firmware:/src/firmware/:rw \
-v /tmp/.X11-unix:/tmp/.X11-unix:ro \
-e DISPLAY=:0 \
-p 14556:14556/udp \
--name=mycontainer px4io/px4-dev-ros:2017-10-23 bash
```

dagar, 2018-03-21:
If xhost is needed you could add it to the Dockerfiles. https://github.com/PX4/containers/blob/master/docker/px4-dev/Dockerfile_base

margaridaCF, 2018-07-25:
Hi! This is a bit late but if anyone bumps into this issue, to accomplish this run xhost+ outside the container before you run it.

closed 2018-08-04 by dagar

NimmiW, 2019-03-21:
Does anyone has an answer for this?