PX4/PX4-containers #156: gst_camera_plugin problem on newer px4io/px4-dev-ros

ArkadiuszNiemiec, 2019-01-18:
I have tried to run a simulation with `gst_camera_plugin` based on `px4io/px4-dev-ros:2019-01-17` and ended up with while building the firmware:
```
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
GSTREAMER_LIBRARIES (ADVANCED)
```
After trying to install gstreamer as described in [https://dev.px4.io/en/simulation/gazebo.html](https://dev.px4.io/en/simulation/gazebo.html):
```
sudo apt-get install $(apt-cache --names-only search ^gstreamer1.0-* | awk '{ print $1 }' | grep -v gstreamer1.0-hybris) -y
```
...I got this error:
```
gstreamer1.0-plugins-base is already the newest version (1.14.1-1ubuntu1~ubuntu18.04.1).
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 gstreamer1.0-python3-dbg-plugin-loader : Conflicts: gstreamer1.0-python3-plugin-loader but 1.14.1-1~ubuntu18.04.1 is to be installed
 gstreamer1.0-python3-plugin-loader : Conflicts: gstreamer1.0-python3-dbg-plugin-loader but 1.14.1-1~ubuntu18.04.1 is to be installed
E: Unable to correct problems, you have held broken packages.
```

On the default tag of `px4io/px4-dev-ros`, which is `px4io/px4-dev-ros:latest`, I am able to run `sudo apt-get install $(apt-cache --names-only search ^gstreamer1.0-* | awk '{ print $1 }' | grep -v gstreamer1.0-hybris) -y` and use this plugin. I think that you have recently changed the version of Ubuntu or something different that created this problem.

I want to use newer `px4io/px4-dev-ros` because it's using a ros melodic instead of kinetic.

AlexLuya, 2019-03-22:
I have exact same problem,have you solved this?

ArkadiuszNiemiec, 2019-03-25:
I have wrote my own Dockerfile with:
```
# --- gstreamer deps ---
RUN ["/bin/bash", "-c", "\
    apt update && \
    apt-get install $(apt-cache --names-only search ^gstreamer1.0-* | awk '{ print $1 }' | grep -v gstreamer1.0-hybris) -y \
"]
```

BazookaJoe1900, 2019-04-26:
@ArkadiuszNiemiec did you make it to work?

ArkadiuszNiemiec, 2019-04-29:
@BazookaJoe1900 I have made a custom Dockerfile that runs this:
```
# --- gstreamer deps ---
RUN ["/bin/bash", "-c", "\
    apt update && \
    apt-get install $(apt-cache --names-only search ^gstreamer1.0-* | awk '{ print $1 }' | grep -v gstreamer1.0-hybris) -y \
"]
```
before cloning and building PX4 Firmware.

Terrangerratt, 2019-05-06:
@BazookaJoe1900 I'm running into the same issue. Could you explain how to use your docker file (sorry I'm new to this) and what the docker file provides for us?

TSC21, 2019-06-27:
@ArkadiuszNiemiec is this still a problem for you?

RyanHurst, 2019-07-17:
I ran into this same problem when following the instructions to enable video stream in gazebo
https://dev.px4.io/v1.9.0/en/simulation/gazebo.html#video-streaming

hamishwillee, 2019-07-17:
@TSC21 Et al, if this ends up requiring a docs fix, can you please keep me posted (would not like to miss it).

Ahmad-Drak, 2019-07-18:
> I ran into this same problem when following the instructions to enable video stream in gazebo
> https://dev.px4.io/v1.9.0/en/simulation/gazebo.html#video-streaming

Same here

bozkurthan, 2019-08-02:
Hello, @hamishwillee Is there any updates on docs for this issue  :eyes:

hamishwillee, 2019-08-02:
@bozkurthan Not that I know of, and I am not tracking it. I am hoping @RyanHurst @TSC21 will ping me if there is any doc change required.

julianoes, 2020-01-10:
It looks like the gstreamer dependencies are now installed in the simulation containers:

https://github.com/PX4/containers/blob/e0d8c0a211c97c4fd73f565052d54d549c73397b/docker/Dockerfile_simulation-bionic#L15-L21

https://github.com/PX4/containers/blob/e0d8c0a211c97c4fd73f565052d54d549c73397b/docker/Dockerfile_simulation-xenial#L15-L21

closed 2020-01-10 by julianoes