PX4/PX4-containers #160: Archlinux container broken

TSC21, 2019-01-24:
According to @julianoes, may be related to ttps://bugs.archlinux.org/task/61428.

lamping7, 2019-03-08:
I manually triggered it in docker hub and it built. The base container must have been fixed. I think we need to not use their `latest` tag for the base here. Thoughts @julianoes ? Should we pin it down to a particular tag?

julianoes, 2019-03-12:
@lamping7 I think lets rebuild using `latest`, yes.

lamping7, 2019-03-12:
I meant change this line to a tag other than `latest`:
https://github.com/PX4/containers/blob/c4943a88c81210162d0a29a354c901fe56233832/docker/px4-dev/Dockerfile_base-archlinux#L5

I looked at their repo and they don't push versioned tags, which is unfortunate. So, in order to prevent random breaks like this, we wound need to push our own versioned tags to our repo. I'm not sure if anyone wants to maintain this though.

So, rebuilding it did work fine. We can probably re-enable the Jenkins test for now and close this, unless we want to do something about the tagging issue.

TSC21, 2019-06-27:
@julianoes @lamping7 does it make sense trying to reboot the ArchLinux container at this stage?

lamping7, 2019-06-27:
Sure. Give it a shot.

TSC21, 2019-08-02:
Solved now.

closed 2019-08-02 by TSC21