PX4/PX4-containers #187: Discussion: Push Jenkins build to Docker Hub

lamping7, 2019-05-22:
Any thoughts on pushing the Jenkins build to Docker Hub rather than tagging and having Docker Hub re-build? We could still tag the new images with the date.

Opening discussion... @dagar @TSC21

dagar, 2019-05-22:
Initially it was just easier to have docker hub do it, but now we have Jenkins working where we can hold the credentials.

julianoes, 2020-01-10:
I've disabled the DockerHub auto builds and it's now doing the `docker build` using GitHub actions:
https://github.com/PX4/containers/actions

It's much quicker because there it runs faster and in parallel.

I removed the pull request checks because they did not really work anyway if you updated the docker tags to the latest date/tag which is not built yet when in a pull request. Also, the CI builds often had circular dependencies and would fail anyway.

closed 2020-01-10 by julianoes