PX4/PX4-containers #188: px4io/px4-dev-ros-kinetic catkin build failed

TianmingLi, 2019-05-23:
- pull docker image 
`docker pull px4io/px4-dev-ros-kinetic`

- create container 

```
docker run -it --privileged \
    -v /home/a:/root/ros_dev:rw \
    -v /tmp/.X11-unix:/tmp/.X11-unix:ro \
    -e DISPLAY=$DISPLAY \
    --name=container_name px4io/px4-dev-ros-kinetic /bin/bash
```

- login to container 
`docker run -it  --user root dev_ros_kinetic `

- create ros package through catkin
`catkin_create_pkg beginer std_msgs --rosdistro Kinetic`

- build
`catkin build` 

- error is 
```
root@48ad1e172536:/catkin_ws# catkin build
-----------------------------------------------
Profile:                     default
Extending:                   None
Workspace:                   /catkin_ws
-----------------------------------------------
Source Space:       [exists] /catkin_ws/src
Log Space:          [exists] /catkin_ws/logs
Build Space:        [exists] /catkin_ws/build
Devel Space:        [exists] /catkin_ws/devel
Install Space:      [unused] /catkin_ws/install
DESTDIR:            [unused] None
-----------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
-----------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-----------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-----------------------------------------------

-----------------------------------------------
WARNING: Your workspace is not extending any
other result space, but it is set to use a
`linked` devel space layout. This requires the
`catkin` CMake package in your source space in
order to be built.
-----------------------------------------------

[build] Found '1' packages in 0.0 seconds.                                                                         
[build] Package table is up to date.                                                                               
Warning: generated devel space setup files have been deleted.
Starting  >>> catkin_tools_prebuild                                                                                
___________________________________________________________________________________________________________________
Errors     << catkin_tools_prebuild:cmake /catkin_ws/logs/catkin_tools_prebuild/build.cmake.002.log                
CMake Error at /catkin_ws/build/catkin_tools_prebuild/CMakeLists.txt:12 (message):
  The catkin CMake module was not found, but it is required to build a linked
  workspace.  To resolve this, please do one of the following, and try
  building again.

   1. Source the setup.sh file from an existing catkin workspace:
      source SETUP_FILE

   2. Extend another catkin workspace's result (install or devel) space:
      catkin config --extend RESULT_SPACE

   3. Set `catkin_DIR` to the directory containing `catkin-config.cmake`:
      catkin config --cmake-args -Dcatkin_DIR=CATKIN_CMAKE_CONFIG_PATH

   4. Add the catkin source package to your workspace's source space:
      cd SOURCE_SPACE && git clone https://github.com/ros/catkin.git

cd /catkin_ws/build/catkin_tools_prebuild; catkin build --get-env catkin_tools_prebuild | catkin env -si  /usr/bin/cmake /catkin_ws/build/catkin_tools_prebuild --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/catkin_ws/devel/.private/catkin_tools_prebuild -DCMAKE_INSTALL_PREFIX=/catkin_ws/install; cd -
...................................................................................................................
Failed     << catkin_tools_prebuild:cmake          [ Exited with code 1 ]                                          
Failed    <<< catkin_tools_prebuild                [ 0.0 seconds ]                                                 
Abandoned <<< beginer                              [ Unrelated job failed ]                                        
[build] Summary: 0 of 2 packages succeeded.                                                                        
[build]   Ignored:   None.                                                                                         
[build]   Warnings:  None.                                                                                         
[build]   Abandoned: 1 packages were abandoned.                                                                    
[build]   Failed:    1 packages failed.                                                                            
[build] Runtime: 0.0 seconds total.    

```

TianmingLi, 2019-05-23:
after 
> Initialize rosdep

and

> Environment setup

http://wiki.ros.org/kinetic/Installation/Ubuntu

catkin build passed

TSC21, 2019-05-23:
We are not sourcing the ROS environment on the container. That has to be something done by the user manually or by adding it to the entrypoint.

closed 2019-05-23 by TSC21