PX4/PX4-containers #195: Arch Script 404 and no SITL

Pedro-Roque, 2019-07-09:
Explained in the title:
https://github.com/PX4/containers/blob/master/docker/px4-dev/scripts/archlinux_install_script.sh

Also, the Docs don't contain simulator setup for Gazebo (at least)
@dagar is the script out of reach outside docker? 

How can I help with this?

TSC21, 2019-07-09:
@Pedro-Roque those scripts are not available for some time now. But we do have an archlinux container available? What are you looking for exactly?

Pedro-Roque, 2019-07-09:
@TSC21 essentially an easy script setup for Arch users, such that we can just run one script to install everything (like in Ubuntu). The container for Arch Linux contains all tools and simulators? Is this an Arch container or a container *for* Arch with the Ubuntu setup?

After setting up my system on Arch, I figured that maybe the same scripts would be of interest to other users.

TSC21, 2019-07-09:
I would say that scripts for those kind of setups are not supposed to be in this repo but rather on the devguide. So this is something to bring to the devguide instead of being opened here.

Pedro-Roque, 2019-07-09:
I did so, ref here:  https://github.com/PX4/Devguide/pull/832#issuecomment-509128726

hamishwillee, 2019-07-09:
@TSC21 We are moving scripts out of devguide into the Firmware tree and in use from Docker - allowing them to be regularly tested. For example, here is the setup for Ubuntu: https://github.com/PX4/Firmware/blob/master/Tools/setup/ubuntu.sh

So while https://github.com/PX4/containers/blob/master/docker/px4-dev/scripts/archlinux_install_script.sh might not exist, we should have a script that does this "somewhere" that is well tested.
Thoughts?

hamishwillee, 2019-07-09:
@Pedro-Roque We probably should close this if the scripts are no longer alive/maintained. Perhaps recreate in Firmware/Tools/setup/archlinux.sh . Let's see what @dagar suggests.

Pedro-Roque, 2019-07-09:
> @Pedro-Roque We probably should close this if the scripts are no longer alive/maintained. Perhaps recreate in Firmware/Tools/setup/archlinux.sh . Let's see what @dagar suggests.

But I do agree that the scripts should be regularly tested, especially with Arch being a rolling distro.

hamishwillee, 2019-07-09:
Yes. It doesn't matter so much where they are as long as the setup is in CI. But I like the idea of having all setup scripts for the environment living together. That implies Firmware/Tools/setup

dagar, 2019-07-10:
Placing it in https://github.com/PX4/Firmware/tree/master/Tools/setup would be great. Then we can be sure developers, documentation, and CI are as in sync as they're going to get. Ideally the script should be idempotent so that it's safe to periodically run on an existing setup to grab any changes.

hamishwillee, 2019-07-10:
@Pedro-Roque Are you happy to submit a script there? - Based on the "design" of https://github.com/PX4/Firmware/blob/master/Tools/setup/ubuntu.sh

Pedro-Roque, 2019-07-10:
@dagar @hamishwillee Yep that seems to be a good solution. Just a question, do we need to setup an Arch docker, or will the script run by default in one?

julianoes, 2020-01-10:
We now have and install script for Arch Linux: https://github.com/PX4/Firmware/blob/master/Tools/setup/arch.sh

However, we still need to use the these docker containers to use the install scripts.

closed 2020-01-10 by julianoes