PX4/PX4-containers #198: Running Gazebo with camera plugin on headless machine

urock, 2019-08-15:
Hi guys!
I am trying to setup CI for px4 + ROS simulation in Gazebo with on-board depth camera. My platform is standalone Ubuntu 16.04 machine without monitor but with nvidia card and I use ROS Kinetic with Gazebo 7. No Docker. 

First I want to run simulation over ssh connection. Basic simulation works fine, but I am having hard time getting to work depth camera. Gazebo output smth like this
```
(1565868372 991876482) [Err] [RenderEngine.cc:699] Can't open display
(1565868372 991904052) [Wrn] [RenderEngine.cc:97] Unable to create X window. Rendering will be disabled
(1565868372 991936077) [Wrn] [RenderEngine.cc:301] Cannot initialize render engine since render path type is NONE. Ignore this warning ifrendering has been turned off on purpose.
(1565868373 531271974) [Err] [DepthCameraSensor.cc:82] Unable to create DepthCameraSensor. Rendering is disabled.
```

After searching internet I found one solution
http://answers.gazebosim.org/question/14625/running-a-camera-sensor-headless/
which is to use xvbf 
```
$ Xvfb :1 -screen 0 1600x1200x16  &
$ export DISPLAY=:1.0
$ roslaunch etc etc.launch
```
also I have noticed that you use the same in your docker setup
https://github.com/PX4/containers/blob/master/docker/px4-dev/scripts/entrypoint.sh

But this solution doesn't work for me, after I give such commands Gazebo fails to start with error
```
> $ gzserver                                                                                       
X Error of failed request:  BadValue (integer parameter out of range for operation)
  Major opcode of failed request:  151 (GLX)
  Minor opcode of failed request:  3 (X_GLXCreateContext)
  Value in failed request:  0x0
  Serial number of failed request:  18
  Current serial number in output stream:  19
```
I think that might be somehow related to nvidia driver. 

Sorry for offtopic but any suggestions would be very much appreciated.

julianoes, 2020-01-10:
@urock were you able to fix the issue in the meantime? When on nvidia I suggest to try using: https://github.com/NVIDIA/nvidia-docker

julianoes, 2020-01-10:
I'm going to close this issue. Please re-open if you think there is something specific to fix in this repo. Thanks!

closed 2020-01-10 by julianoes

JBLanier, 2020-03-09:
Hi was this issue ever solved? I am having the exact same issue.

hamishwillee, 2020-03-09:
Nope, it was closed. Suggest you recreate if you think this is a PX4 issue.