PX4/PX4-containers #266: Using container on RaspberryPi

MaximeAubanel, 2020-06-03:
Hello,

I need px4io/px4-dev-ros2-eloquent to run on RaspberryPi or ARMv7 architecture. Can you provide it to us please ? If not how can we do it ?

TSC21, 2020-06-03:
@MaximeAubanel these images are aimed for usage with CI pipelines, they are optimised for deployment. So you will have to dig in further and create your own container. But I am not really sure you are actually able to do that since https://registry.hub.docker.com/r/armv7/armhf-ubuntu only has images until Ubuntu 16.10, and you need at least 18.04.

I would start by probably first understand how to install it natively (http://www.lib4dev.in/info/akash-roboticist/rpi-ros2/230350899) and then understand how can that be brought to a container.

MaximeAubanel, 2020-06-03:
Thanks for that blazing fast answer.

But let say I want to communicate between my ROS1 program and my drone. Then at some point, I need to have mavros launched connected to the drone using the `fcu_url` right ? Then it means that on my embedded device I'm going to require mavros and the whole px4 toolchain no ?

Or am I completely wrong ? If I am, what image would I need to run `roslaunch mavros px4.launch fcu_url:="udp://:14540@host.docker.internal:14557"` ?

TSC21, 2020-06-03:
> Thanks for that blazing fast answer.
>
> But let say I want to communicate between my ROS1 program and my drone. Then at some point, I need to have mavros launched connected to the drone using the [...]

I am not getting it here: if you are to use ROS1, why do you want ROS2?

MaximeAubanel, 2020-06-03:
Oh yeah my bad I didn't mention this. We are using both ROS1 and ROS2 on our nodes, _linked_ with the bridge.

TSC21, 2020-06-03:
Just in any case, we don't have any image on this container for ARM that has ROS1/2 inside. It will have to be something you need to build yourselves. I advise you look for a docker image for ARMv7 and then create a Dockerfile that extends it with ROS and the required dependencies.

TSC21, 2020-06-03:
You can start by having a look at this thread: https://discourse.ros.org/t/announcing-ros-docker-images-for-arm-and-debian/2467.

MaximeAubanel, 2020-06-03:
Okay. Thanks for your answer. 

As ARM 32 (armv7) doesn't support Ubuntu 18.04 and Ubuntu 18.04 is required I will have to switch to an ARM 64 device.

Let's say,  I isolate my ROS1 program, even using px4io/px4-dev-ros-melodic on an ARM 64 device won't work because it's not made for ARM 64 architecture. So basically it means that I need to build the image myself. Right ?

If it is you can close the issue.

closed 2020-09-01 by MaximeAubanel