PX4/PX4-containers #289: Inaccurate Documentation

aseyfi, 2021-02-03:
Hello,

I was following the instructions in [this link](https://docs.px4.io/master/en/test_and_ci/docker.html#container-hierarchy). Looking at this table my understanding is that "px4-dev-ros" can do everything "px4-dev-nuttx" does, plus extra ROS and simulation applications, but it is not the case. "px4-dev-ros" doesn't include arm tool-chain and nuttx environment setup.

I am new to PX4 and Docker, so please forgive me, if this is a false alarm or this git repository is not the right place.

Thanks,
Ahmad

hamishwillee, 2021-02-04:
Hi @aseyfi Thanks for posting.

This issue should be in the docs repo: https://github.com/PX4/PX4-user_guide

@TSC21 The table in https://docs.px4.io/master/en/test_and_ci/docker.html#container-hierarchy seems not to match well with https://github.com/PX4/PX4-containers#container-hierarchy
Any thoughts on what should be in the docs?

TSC21, 2021-02-04:
> Hi @aseyfi Thanks for posting.
>
> This issue should be in the docs repo: https://github.com/PX4/PX4-user_guide
>
> @TSC21 The table in [...]

The docs needs to be in sync with what is in here or just be removed.

hamishwillee, 2021-02-10:
Thanks. Fixed by https://github.com/PX4/PX4-user_guide/pull/1045

closed 2021-02-10 by hamishwillee