PX4/PX4-containers #316: jenkins use the docker images failed

ShiauweiZhao, 2022-05-04:
When I call docker images with jenkins, he reports the following error
```
+ ccache -s
cache directory                     /tmp/ccache
primary config                      /tmp/ccache/ccache.conf
secondary config      (readonly)    /etc/ccache.conf
cache hit (direct)                     0
cache hit (preprocessed)               0
cache miss                             0
cache hit rate                      0.00 %
cleanups performed                     0
files in cache                         0
cache size                           0.0 kB
max cache size                       5.0 GB
[Pipeline] sh
+ make beaglebone_blue_default
-- PX4 version: v1.13.0-beta1
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.7.3", minimum required is "3") 
-- PX4 config file: /build/jenkins/workspace/px4_beta/boards/beaglebone/blue/default.px4board
-- PLATFORM posix
-- LINUX y
-- TOOLCHAIN arm-linux-gnueabihf
-- ARCHITECTURE cortex-a8
-- ROMFSROOT px4fmu_common
-- TESTING y
-- PX4 config: beaglebone_blue_default
-- PX4 platform: posix
-- PX4 lockstep: disabled
-- cmake build type: RelWithDebInfo
-- The CXX compiler identification is GNU 8.3.0
-- The C compiler identification is GNU 8.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/lib/ccache/arm-linux-gnueabihf-gcc
-- Check for working CXX compiler: /usr/lib/ccache/arm-linux-gnueabihf-g++
-- Check for working CXX compiler: /usr/lib/ccache/arm-linux-gnueabihf-g++ -- broken
CMake Error at /usr/share/cmake-3.13/Modules/CMakeTestCXXCompiler.cmake:45 (message):
  The C++ compiler

    "/usr/lib/ccache/arm-linux-gnueabihf-g++"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /build/jenkins/workspace/px4_beta/build/beaglebone_blue_default/CMakeFiles/CMakeTmp
    
    Run Build Command:"/usr/bin/ninja" "cmTC_239a3"
    [1/2] Building CXX object CMakeFiles/cmTC_239a3.dir/testCXXCompiler.cxx.o
    FAILED: CMakeFiles/cmTC_239a3.dir/testCXXCompiler.cxx.o 
    /usr/lib/ccache/arm-linux-gnueabihf-g++     -o CMakeFiles/cmTC_239a3.dir/testCXXCompiler.cxx.o -c testCXXCompiler.cxx
    ccache: error: Failed to create directory /tmp/ccache/tmp: Permission denied
    ninja: build stopped: subcommand failed.
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:235 (project)

-- Configuring incomplete, errors occurred!
See also "/build/jenkins/workspace/px4_beta/build/beaglebone_blue_default/CMakeFiles/CMakeOutput.log".
See also "/build/jenkins/workspace/px4_beta/build/beaglebone_blue_default/CMakeFiles/CMakeError.log".
Error: /build/jenkins/workspace/px4_beta/build/beaglebone_blue_default is not a directory
make: *** [Makefile:230: beaglebone_blue_default] Error 1
```
However, in github actions he works fine,I also tested docker images myself in pc and it works fine
,I think it's because the way jenkins calls docker images is different from github actions,The situation may look like this link  https://stackoverflow.com/questions/58205082/in-jenkins-how-to-use-docker-image-with-cross-compilation-environment

dagar, 2022-05-04:
How are you launching this? In PX4-Autopilot usage the CCACHE_DIR is passed in as a volume so the cache can be shared between builds.

https://github.com/PX4/PX4-Autopilot/blob/bf8d759d3d497740512428fde82ad756bd5221c0/Tools/docker_run.sh#L65

You probably need to make sure the permissions of your CCACHE_DIR is set appropriately so the docker user can actually write to it.

ShiauweiZhao, 2022-05-05:
> How are you launching this? In PX4-Autopilot usage the CCACHE_DIR is passed in as a volume so the cache can be shared between builds. [...]
I use PX4-AutoPilot's Jenkinsfile-compile to call docker.i'll double check the issue

ShiauweiZhao, 2022-05-05:
close because CCACHE_DIR  not have permissions for jenkins

closed 2022-05-05 by ShiauweiZhao