PX4/PX4-containers #330: Docker image for arm64/apple silicon

kurosh-z, 2023-02-05:
Is there any docker image compatible with apple silicon? I tried px4-dev-noetic:latest pulled form docker hub and I'm getting the following error on ubuntu 20.04 running inside the virtual machine:
[image]

is there any instruction on how to rebuild the images for arm64?

cryptik, 2023-02-20:
I am having a similar issue... but I am trying to build in a docker container on NVIDA Jetson Orin, which is also an aarch64 platform like the M1.

After running either the helper script `./docker_run.sh make px4_sitl jmavsim` or manually with command

'''
docker run -it --privileged \
--env=LOCAL_USER_ID="$(id -u)" \
--platform=linux/amd64 \
-v ~/Documents/tools/PX4-Autopilot:/src/PX4-Autopilot/:rw \
-v /tmp/.X11-unix:/tmp/.X11-unix:ro \
-e DISPLAY=:0 \
--network host \
--name=px4-dev px4io/px4-dev-aarch64:2022-08-12 bash
'''
I get `exec /usr/local/bin/entrypoint.sh: exec format error` when the container attempts to run the entrypoint script.  Various things on the web indicate adding `export DOCKER_DEFAULT_PLATFORM=linux/amd64` to the `.bashrc` file, which I did, but I still get the error.  After setting the environment var above, I deleted the container and the image and re-ran the commands but it still generates the format error.  I don't know enough about docker and the entrypoint script to know what to do next.

I am guessing that either `exec gosu user bash` or this `exec  bash` is where the problem is as it is trying to run `bash` on the `aarch64` hardware and the executable is `amd64`.  However, it was my understanding that setting the platform to `linux/amd64` would force docker to run the container in emulation mode on the `aarch64` hardware, but I am not sure.

Is it not possible at this point to build the PX4 simulator natively on the Orin or any linux arm?

kurosh-z, 2023-02-21:
I could build the docker images by applying the following changes:

**in the Dockerfile_base-focal :**
```

ENV DEBIAN_FRONTEND noninteractive
ENV LANG C.UTF-8
ENV LC_ALL C.UTF-8

RUN apt-get update && apt-get -y --quiet --no-install-recommends install \
		bzip2 \
		ca-certificates \
		ccache \
		cmake \
		cppcheck \
		curl \
		dirmngr \
		doxygen \
		file \Dockerfile_base-focal
		g++ \
		g++-aarch64-linux-gnu  \
		gcc \
		gcc-aarch64-linux-gnu \
		gdb \
		git \
		gnupg \
		gosu \
		lcov \
		libfreetype6-dev \
		libgtest-dev \
		libpng-dev \
		libssl-dev \
		lsb-release \
		make \
		ninja-build \
		openjdk-8-jdk \
		openjdk-8-jre \
		openssh-client \
		pkg-config \
		python3-dev \
		python3-pip \
		rsync \
		shellcheck \
		tzdata \
		unzip \
		valgrind \
		wget \
		xsltproc \
		zip \
		nano \
		vim \
	&& apt-get -y autoremove \
	&& apt-get clean autoclean \
	&& rm -rf /var/lib/apt/lists/{apt,dpkg,cache,log} /tmp/* /var/tmp/*

```

**in the Dockerfile_simulation-focal:**
I couldn't find the JSBSim build for arm64 so I disabled it:
```
# Install JSBSim
# RUN wget https://github.com/JSBSim-Team/jsbsim/releases/download/v1.1.1a/JSBSim-devel_1.1.1-134.focal.amd64.deb
# RUN dpkg -i JSBSim-devel_1.1.1-134.focal.amd64.deb

```

**in the Dockerfile_ros-noetic:**
```
#
# PX4 ROS development environment
#

FROM px4io/px4-dev-simulation-focal-arm64:latest
LABEL maintainer="Nuno Marques <n.marques21@hotmail.com>"

ENV ROS_DISTRO noetic

# setup ros keys
RUN curl -s https://raw.githubusercontent.com/ros/rosdistro/master/ros.asc | apt-key add -

RUN sh -c 'echo "deb http://packages.ros.org/ros/ubuntu `lsb_release -sc` main" > /etc/apt/sources.list.d/ros-latest.list' \
	&& sh -c 'echo "deb http://packages.ros.org/ros/ubuntu `lsb_release -sc` main" > /etc/apt/sources.list.d/ros-latest.list' \
	&& sh -c 'echo "deb http://packages.ros.org/ros-shadow-fixed/ubuntu `lsb_release -sc` main" > /etc/apt/sources.list.d/ros-shadow.list' \
	&& apt-get update \
	&& apt-get -y --quiet --no-install-recommends install \
		geographiclib-tools \
		libeigen3-dev \
		libgeographic-dev \
		libopencv-dev \
		libyaml-cpp-dev \
		python3-rosdep \
		python3-catkin-tools \
		python3-catkin-lint \
		ros-$ROS_DISTRO-gazebo-ros-pkgs \
		ros-$ROS_DISTRO-mavlink \
		ros-$ROS_DISTRO-mavros \
		ros-$ROS_DISTRO-mavros-extras \
		ros-$ROS_DISTRO-octomap \
		ros-$ROS_DISTRO-octomap-msgs \
		ros-$ROS_DISTRO-pcl-conversions \
		ros-$ROS_DISTRO-pcl-msgs \
		ros-$ROS_DISTRO-pcl-ros \
		ros-$ROS_DISTRO-ros-base \
		ros-$ROS_DISTRO-rostest \
		ros-$ROS_DISTRO-rosunit \
		xvfb \
	&& geographiclib-get-geoids egm96-5 \
	&& apt-get -y autoremove \
	&& apt-get clean autoclean \
	&& rm -rf /var/lib/apt/lists/{apt,dpkg,cache,log} /tmp/* /var/tmp/*

RUN pip3 install -U \
		osrf-pycommon

# bootstrap rosdep
RUN rosdep init && rosdep update

```

I then added the following lines in CMake to build the images (make sure that you remove the images you might already download from the docker repo)

```
px4-dev-ros-noetic-arm64: px4-dev-simulation-focal-arm64
	docker build --platform=linux/arm64  -t px4io/px4-dev-ros-noetic-arm64 . -f Dockerfile_ros-noetic
	
px4-dev-simulation-focal-arm64: px4-dev-base-focal-arm64
	docker build --platform=linux/arm64 -t px4io/px4-dev-simulation-focal-arm64 . -f Dockerfile_simulation-focal

px4-dev-base-focal-arm64:
	docker build --platform=linux/arm64 -t px4io/px4-dev-base-focal-arm64 . -f Dockerfile_base-focal

```
build the base-focal-arm64 then px4-dev-simulation-focal-arm64 and finally the px4-dev-ros-noetic-arm64.

cryptik, 2023-02-22:
@kurosh-z thanks a million for this info. I will give this a try. I will give this a try ASAP.